InternRobotics/InternNav #25: No such file or directory: 'data/datasets/navdp_dataset_lerobot.json'

mscwu, 2025-08-16:
I am trying to train NavDP modeling using InternNav framework.

Where do I find this 

> navdp_dataset_lerobot.json

 file required in InternNav/scripts/train/configs/navdp.py? I have checked the repo and InternData-N1 and InternScenes. None of them contains this file. 

https://github.com/InternRobotics/InternNav/blob/8ef766d16a34f08d9a6d5e666a57caefe5b9b58a/scripts/train/configs/navdp.py#L53

And when I run `./scripts/train/start_train.sh --name train_navdp --model navdp`, it runs into this error: No such file or directory: 'data/datasets/navdp_dataset_lerobot.json'

shuishui602, 2025-11-13:
？

wzcai99, 2025-11-13:
By setting config file preload = False, the dataloader will automatically generate this file.