InternRobotics/InternNav #50: Error of the evaluation.

LuoXubo, 2025-09-01:
Hi. I followed the README in `challenge` to evaluate the baseline model. I've downloaded the datasets and baseline models. However, when I run the following command:
```
$ ./scripts/eval/start_eval.sh --config scripts/eval/configs/challenge_cfg.py
```
there is something wrong:
```
(internutopia) root@xubo-ubuntu:~/InternNav# ./scripts/eval/start_eval.sh --config scripts/eval/configs/challenge_cfg.py
Error while loading conda entry point: conda-anaconda-tos (cannot import name 'BaseModel' from 'pydantic' (/root/miniconda3/lib/python3.13/site-packages/pydantic/__init__.py))
kill: 730
Starting process...
start healthcheck
Process 943 has exited.
Retrying... (Attempt 1/5)
Starting process...
start healthcheck
Process 1014 has exited.
Retrying... (Attempt 2/5)
Starting process...
start healthcheck
Process 1050 has exited.
Retrying... (Attempt 3/5)
Starting process...

```
Could you help me solve this problem?

kew6688, 2025-09-02:
Could you please provide more details from `logs/`?

LuoXubo, 2025-09-02:
Sure. 
```
(base) root@xubo-ubuntu:~/InternNav# ./scripts/eval/start_eval.sh --config scripts/eval/configs/challenge_cfg.py
Starting process...
start healthcheck
Process 527 has exited.
Retrying... (Attempt 1/5)
Starting process...
start healthcheck
Process 874 has exited.
Retrying... (Attempt 2/5)
Starting process...
start healthcheck
Process 1179 has exited.
Retrying... (Attempt 3/5)
Starting process...
start healthcheck
Process 1479 has exited.
Retrying... (Attempt 4/5)
Starting process...
start healthcheck
Process 1783 has exited.
Retrying... (Attempt 5/5)
Starting process...
start healthcheck
Process 2091 has exited.
Exceeded maximum retry attempts. Exiting.
```
The `challenge_eval.log` in `logs/` has the following errors:
```
[7.986s] Simulation App Startup Complete
[2025-09-02 05:23:09,472][DEBUG] /InternUtopia/internutopia/core/runner.py[line:488] -: SimulationApp init done
[2025-09-02 05:23:09,472][INFO] /InternUtopia/internutopia/core/runner.py[line:467] -: simulator params: physics dt=0.005, rendering dt=0.005, use_fabric=False
2025-09-02 05:23:09 [7,990ms] [Warning] [omni.isaac.core.scenes.scene] omni.isaac.core.scenes.scene has been deprecated in favor of isaacsim.core.api.scenes.scene. Please update your code accordingly.
2025-09-02 05:23:09 [7,991ms] [Warning] [omni.isaac.core.scenes.scene_registry] omni.isaac.core.scenes.scene_registry has been deprecated in favor of isaacsim.core.api.scenes.scene_registry. Please update your code accordingly.
[2025-09-02 05:23:09,486][INFO] /InternUtopia/internutopia/core/runner.py[line:47] -: rendering interval: 5
[2025-09-02 05:23:09,488][DEBUG] /isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connectionpool.py[line:243] -: Starting new HTTP connection (1): localhost:8087
[2025-09-02 05:23:21,777][DEBUG] /isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connectionpool.py[line:546] -: http://localhost:8087 "POST /agent/init HTTP/11" 500 21
2025-09-02 05:23:21 [20,282ms] [Error] [omni.kit.app._impl] [py stderr]: Traceback (most recent call last):
  File "/root/InternNav/scripts/eval/eval.py", line 38, in <module>
    main()
  File "/root/InternNav/scripts/eval/eval.py", line 33, in main
    evaluator = Evaluator.init(cfg)
  File "/root/InternNav/internnav/evaluator/base.py", line 37, in init
    return cls.evaluators[config.env.env_type](config)
  File "/root/InternNav/internnav/evaluator/vln_pe_evaluator.py", line 81, in __init__
    super().__init__(config)
  File "/root/InternNav/internnav/evaluator/base.py", line 16, in __init__
    self.agent = AgentClient(config.agent)
  File "/root/InternNav/internnav/agent/utils/client.py", line 16, in __init__
    self.agent_name = self._initialize_agent(config)
  File "/root/InternNav/internnav/agent/utils/client.py", line 26, in _initialize_agent
    response.raise_for_status()
  File "/isaac-sim/exts/omni.pip.cloud/pip_prebundle/requests/models.py", line 1024, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 500 Server Error: Internal Server Error for url: http://localhost:8087/agent/init
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-09-02 05:23:21 [20,304ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-09-02 05:23:21 [20,309ms] [Warning] [omni.graph.core.plugin] Could not find category 'animation' for removal
2025-09-02 05:23:21 [20,310ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-09-02 05:23:21 [20,310ms] [Warning] [carb.audio.context] 1 contexts were leaked
2025-09-02 05:23:21 [20,358ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

kew6688, 2025-09-02:
It looks like the model didn't init properly. Could you share the challenge_server.log as well?

LuoXubo, 2025-09-02:
Yes:
```
(base) root@xubo-ubuntu:~/InternNav# bash challenge/start_eval_iros.sh --config scripts/eval/configs/challenge_cfg.py
Usage stats disabled for future clusters. Restart any current running clusters for this to take effect.
Did not find any active Ray processes.
Usage stats collection is disabled.

Local node IP: 10.2.29.219
2025-09-02 06:42:14,173	WARNING services.py:2152 -- WARNING: The object store is using /tmp instead of /dev/shm because /dev/shm has only 67108864 bytes available. This will harm performance! You may be able to free up space by deleting files in /dev/shm. If you are inside a Docker container, you can increase /dev/shm size by passing '--shm-size=10.24gb' to 'docker run' (or add it to the run_options list in a Ray cluster config). Make sure to set this to more than 30% of available RAM.

--------------------
Ray runtime started.
--------------------

Next steps
  To add another node to this Ray cluster, run
    ray start --address='10.2.29.219:6379'
  
  To connect to this Ray cluster:
    import ray
    ray.init()
  
  To terminate the Ray runtime, run
    ray stop
  
  To view the status of the cluster, use
    ray status
Starting process...
```
`eval_stderr.log`:
```
[2025-09-02 06:42:21,765][INFO] /root/InternNav/internnav/evaluator/vln_pe_evaluator.py[line:51] -: start eval dataset: challenge_mp3d_eval, total_path:2617
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'any' as 'Any'
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'bool' as 'Bool' and array 'BoolArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'bundle' as 'Bundle'
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'colord[3]' as 'Color3d' and array 'Color3dArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'colorf[3]' as 'Color3f' and array 'Color3fArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'colorh[3]' as 'Color3h' and array 'Color3hArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'colord[4]' as 'Color4d' and array 'Color4dArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'colorf[4]' as 'Color4f' and array 'Color4fArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'colorh[4]' as 'Color4h' and array 'Color4hArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'double' as 'Double' and array 'DoubleArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'double[2]' as 'Double2' and array 'Double2Array
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'double[3]' as 'Double3' and array 'Double3Array
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'double[4]' as 'Double4' and array 'Double4Array
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'execution' as 'Execution'
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'float' as 'Float' and array 'FloatArray
[2025-09-02 06:42:25,456][DEBUG] /isaac-sim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/autonode_generator/ogn_types.py[line:51] -: Defining data type 'float[2]' as 'Float2' and a [... cut ...]
2025-09-02 06:43:09 [20,887ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-09-02 06:43:09 [20,892ms] [Warning] [omni.graph.core.plugin] Could not find category 'animation' for removal
2025-09-02 06:43:09 [20,893ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-09-02 06:43:09 [20,894ms] [Warning] [carb.audio.context] 1 contexts were leaked
2025-09-02 06:43:09 [21,007ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

LuoXubo, 2025-09-02:
Sure,
```
WARNING:  Current configuration will not reload as not all conditions are met, please refer to documentation.
INFO:     Started server process [256]
INFO:     Waiting for application startup.
INFO:     Application startup complete.
INFO:     Uvicorn running on http://localhost:8087 (Press CTRL+C to quit)
/root/InternNav/internnav/model/basemodel/LongCLIP/model/longclip.py:68: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  state_dict = torch.load(model_path, map_location="cpu")
/root/InternNav/internnav/model/encoder/resnet_encoders.py:46: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  ddppo_weights = torch.load(checkpoint)
PROJECT_ROOT_PATH:/root/InternNav
INFO:     127.0.0.1:52994 - "POST /agent/init HTTP/1.1" 500 Internal Server Error
ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connection.py", line 199, in _new_conn
    sock = connection.create_connection(
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/util/connection.py", line 85, in create_connection
    raise err
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
OSError: [Errno 101] Network is unreachable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connectionpool.py", line 789, in urlopen
    response = self._make_request(
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connectionpool.py", line 490, in _make_request
    raise new_e
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connectionpool.py", line 466, in _make_request
    self._validate_conn(conn)
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connectionpool.py", line 1095, in _validate_conn
    conn.connect()
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connection.py", line 693, in connect
    self.sock = sock = self._new_conn()
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connection.py", line 214, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x71fcf8fa7190>: Failed to establish a new connection: [Errno 101] Network is unreachable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/isaac-sim/exts/omni.pip.cloud/pip_prebundle/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/connectionpool.py", line 843, in urlopen
    retries = retries.increment(
  File "/isaac-sim/extscache/omni.kit.pip_archive-0.0.0+d02c707b.lx64.cp310/pip_prebundle/urllib3/util/retry.py", line 519, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='huggingface.co', port=443): Max retries exceeded with url: /roberta-base/resolve/main/config.json (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x71fcf8fa7190>: Failed to establish a new connection: [Errno 101] Network is unreachable'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... cut ...]
    file = cached_files(path_or_repo_id=path_or_repo_id, filenames=[filename], **kwargs)
  File "/root/miniconda3/envs/internutopia/lib/python3.10/site-packages/transformers/utils/hub.py", line 491, in cached_files
    raise OSError(
OSError: We couldn't connect to 'https://huggingface.co' to load the files, and couldn't find them in the cached files.
Checkout your internet connection or see how to run the library in offline mode at 'https://huggingface.co/docs/transformers/installation#offline-mode'.
```
Looks like a connection error.

kew6688, 2025-09-02:
Yes, that's the issue. OSError: [Errno 101] Network is unreachable
In order to run the model for the first time, we required to pull some model from huggingface. Do you have proxy to access to the huggingface?

LuoXubo, 2025-09-02:
Yes, I have a proxy, but it can be unstable at times. Can I save it locally before the evaluation?

kew6688, 2025-09-02:
Okay, great! The proxy is only needed for the first time to run. Please try if you can pull the model from the huggingface.

LuoXubo, 2025-09-02:
Thanks for your help! I will configure the proxy to download the models.