InternRobotics/InternNav #74: [Question]: habitat 仿真

worker16542, 2025-09-25:
### Question

您好，我想问一下，为什么在habitat仿真中，agent总是低头推理，而需要抬头进行移动呢。

worker16542, 2025-09-26:
> 因为低头能够看清脚下的路，而平视能够看到更多的物体，这样更符合人类直觉。

我又仔细看了habitat的仿真代码，其中input_images为s2的输入，为平视图。look_down_image为s1的输入，为低头。也就是s1和s2的输入视角不一样，所以会有一些疑问。比如看现实世界的视频，并没有抬头低头的动作。

yuqiang-yang, 2025-09-28:
是的，你的理解没错，在habitat评测的时候确实是点pixel的时候是低头，平时走的时候是平视。核心原因是我们发现habitat里面要是只低头的话很多物体看不见，会比较影响VLN的性能。。

真机里面不低头主要是因为，1. 我们发现不低头性能也还不错，可能本身相机够高也能看到比较多的东西   2. 由于真机是异步执行的，机器人会一直走，即使模型输出了低头，那么此时再低头机器人已经不在原本位置了 3. 我们也专门训了一版全部不低头的模型给真机

worker16542, 2025-09-28:
> 因为低头能够看清脚下的路，而平视能够看到更多的物体，这样更符合人类直觉。

> 是的，你的理解没错，在habitat评测的时候确实是点pixel的时候是低头，平时走的时候是平视。核心原因是我们发现habitat里面要是只低头的话很多物体看不见，会比较影响VLN的性能。。
> 
> 真机里面不低头主要是因为，1. 我们发现不低头性能也还不错，可能本身相机够高也能看到比较多的东西 2. 由于真机是异步执行的，机器人会一直走，即使模型输出了低头，那么此时再低头机器人已经不在原本位置了 3. 我们也专门训了一版全部不低头的模型给真机

嗯嗯好的，我明白了谢谢。
我还想请教些问题。在真机测试的代码中，我发现了look_down默认=false，此时S2输出左右或者向下，只有当输出向下的时候（真机没有执行低头动作，而类似标志位），再次调用一次S2，此时才输出像素点。为什么选择这种根据look_down而选择不同的输出呢，而不是所有都使用像素点作为输出呢。然后真机中对于S2来说是运行几次look_down=false直到输出向下运行一次look_down=true，然后继续运行look_down=false的流程吗

worker16542, 2025-09-28:
还有一个就是我看在habitat、iscca sim、真实世界，最后要么S2输出动作，要么S1将轨迹转化为动作，这样做的原因是要统一输出接口吗，或者有别的原因吗

yuqiang-yang, 2025-09-29:
之所以同时存在pixel和离散的旋转，是因为有时候目标确实不在 视野内。比如目标在身后时，此时你无论怎么投都无法把目标点投影到当前图像中